Gadgetoid/WiringPi2-Python #27: Refactor SWIG interface file to import headers rather than duplicate them

willprice, 2015-04-24:
Hi,

Is there any reason why you haven't imported the header files instead of hard coding the different function prototypes (e.g. ignoring some functions that the headers provide). If not I'd be happy to do this, it'd make maintaining and augmenting the project a bit easier!

Cheers

Gadgetoid, 2016-02-16:
It's certainly something that needs doing. The only reason I didn't do it, is that until recently I didn't know it was possible.

Gadgetoid, 2016-02-28:
I looked into it, and importing the header files individually seems to break things horribly. I may have done something wrong, but it seemed like a slippery slope.

I've created a bindings generator which can help with the maintenance problems. This parses the C header files and outputs bindings.i with the declarations it thinks should be included.

See: https://github.com/WiringPi/WiringPi-Python/commit/aee4f5de553b833d97b768d6e620cbad9427bc32

(Note: Might be worth raising a new issue over on the WiringPi/WiringPi-Python repo for ongoing discussion of this)

closed 2016-02-28 by Gadgetoid