Gadgetoid/WiringPi2-Python #35: Python 3 compatability

howitzer-88, 2016-02-28:
Please add a #define for PyUnicode_FromStringAndSize to wiringpi_wrap.c this will allow compatibility with PyString_FromStringAndSize. 

that will fix ImportError: /usr/lib/python3.5/site-packages/wiringpi2-1.1.1-py3.5-linux-armv6l.egg/_wiringpi2.cpython-35m-arm-linux-gnueabihf.so: undefined symbol: PyString_FromStringAndSize 

I have a modified wiringpi_wrap.c if you'd like it.

Gadgetoid, 2016-02-28:
wiringpi_wrap.c is generated entirely by SWIG ( 2.0, or possibly now 3.0 ) so we need to work within the confines of that to make fixes/changes to the (generated) bindings.

I will investigate!

Gadgetoid, 2016-02-28:
This appears to be fixed in: https://github.com/WiringPi/WiringPi-Python/commit/2fb6b4999527f2e491d34a58f7eb87d97ffa050e

closed 2016-02-28 by Gadgetoid