Gadgetoid/WiringPi2-Python #37: WIringPi2-Python crashes with Python3

neuralassembly, 2016-03-25:
# Problem

When WIringPi2-Python is used with Python3, the application crashes with the following errors.

```
ImportError: /usr/local/lib/python3.4/dist-packages/wiringpi2-1.1.1-py3.4-linux-armv7l.egg/_wiringpi2.cpython-34m.so: undefined symbol: PyString_FromStringAndSize
```

# Cause

There is no alias of PyString_FromStringAndSize to PyUnicode_FromStringAndSize in wiringpi_wrap.c.

# Solution

The above problem was solved with the following patch.
If you can accept it, I wiil create a pull request.

```
--- wiringpi_wrap.c     2016-03-25 14:11:17.378634031 +0900
+++ wiringpi_wrap.c.mod 2016-03-25 14:29:46.000000000 +0900
@@ -731,6 +731,7 @@
 #define PyInt_FromSize_t(x) PyLong_FromSize_t(x)
 #define PyString_Check(name) PyBytes_Check(name)
 #define PyString_FromString(x) PyUnicode_FromString(x)
+#define PyString_FromStringAndSize(x, len) PyUnicode_FromStringAndSize(x, len)
 #define PyString_Format(fmt, args)  PyUnicode_Format(fmt, args)
 #define PyString_AsString(str) PyBytes_AsString(str)
 #define PyString_Size(str) PyBytes_Size(str)
```

neuralassembly, 2016-03-30:
Hi, is anyone here?

Gadgetoid, 2016-03-30:
Nope. This repository has been deprecated in favour of: https://github.com/WiringPi/WiringPi-Python

See the note in the readme.

trpster, 2016-04-17:
Any progress on this? I see no note related to this in the readme of the link posted above.

trpster, 2016-04-17:
Never mind; figured it out.

lmckusic, 2016-08-14:
To clarify the steps. Copy the stuff called a patchfile above. Use an editor like vi and put it in a file called patchfile. From the WiringPi2-Python directory where the file named above occurs do the patching process by typing the following command: patch <patchfile ;# Old timey Unix! List the files of the directory and see the patch program saved your original file with the name wiringpi_wrap.c.orig
The preceding steps added the symbol link reported missing. Now you can proceed with sudo python3 setup.py install. Test the installation by running python3 and then > import wiringpi2